TengerTechnologies/Bricklayers #1: Gaps in the outter wall

Fujicsso, 2025-01-22:
[image]

This happened in several spots on my test slice, just wanted to mention this

[image]

TengerTechnologies, 2025-01-22:
Can you send me the gCode file and the Log file (which is in the same directory as the script) ?

Fujicsso, 2025-01-22:
Cant send GCODE here, it's not allowed.

[z_shift_log.txt](https://github.com/user-attachments/files/18512466/z_shift_log.txt)

TengerTechnologies, 2025-01-22:
renaming the gcode file to .txt could work

Tiredmechanic, 2025-01-23:
[BLACK ABS PUMP HOUSING.txt](https://github.com/user-attachments/files/18514599/BLACK.ABS.PUMP.HOUSING.txt)

Just giving a 2nd gcode file for you to look at as im having the same issue so maybe having a 2nd file to compare to the 1st can help narrow it down.

rodrykthingol, 2025-01-23:
Same issue here. Did a couple o random experiments. Change mostly the parameters in prusaslicer. The ones that usually remove these outer wall gaps are turning on z hop (no ramping) and doing external walls first printing. Other times you need to increase the number of perimeters. Looking at the gcode, seems like when its printing some inner wall and moves to ther external perimeter it forgets to change z height.

[Bug.zip](https://github.com/user-attachments/files/18524832/Bug.zip)
[Bug fixed.zip](https://github.com/user-attachments/files/18524831/Bug.fixed.zip)

rodrykthingol, 2025-01-23:
If you see the gcode preview on layer 34 , line 17567 of the gcode. It prints the outer wall of the benchy tub, then detects the inner wall and increases z, then when it goes to outer wall of the outside of the benchy it forgets to reduce the value of z height.

AidanBoyer, 2025-01-26:
I have experienced the same issue of intermittent gaps in the outer wall, and can confirm that slicing the same object in OrcaSlicer while making no changes except switching the wall printing order from Inner/Outer to Outer/Inner appears to fix this.