TengerTechnologies/Bricklayers #2: Error Code 9009 When Slicing on orcaslicer

RnK-BnK, 2025-01-22:
I have applied the script in the post-processing scripts as such:

"C;\Pathtopython\Python.exe" "C:\Pathtocode\bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.1

it spits out this error: 

[image]|

I'm not quite sure how to progress on this and any advise would be ideal, thanks!

DBT85, 2025-01-22:
I get this when I use the "\AppData\Local\Microsoft\WindowsApps\python.exe" path.

If I use the "\AppData\Local\Programs\Python\Python312\python.exe" path I actually get 

"\AppData\Local\Programs\Python\Python312\python.exe"

Post-processing script "C:\Users\yourmother\AppData\Local\Programs\Python\Python312\python.exe"
"‪D:\Sync\3D Print Archive\bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.1 on file
C:\Users\yourmother\AppData\Local\Temp/orcaslicer_model/Wed_Jan_22/21_51_30#15964#138/Meta
data\.15964.1.gcode failed.
Error code: 2

RnK-BnK, 2025-01-22:
I tired switching to using "\AppData\Local\Programs\Python\Python312\python.exe" however when i ran the script and re-imported the g-code it does not seems to have applied teh bricklayers

Ati-93, 2025-01-23:
Hi Folks,

i got the same Issues with Error 9009 and Error 2 in Orca Slicer.
Do you have any Ideas?

natecampbell-zz, 2025-01-23:
This is due to some windows weirdness and I also had this issue. The solution for me was to open PC Settings - Apps - and click on "App execution aliases" and turn them off for the python entries I then did a search for the python.exe which was located in "C:\Users\UserName\AppData\Local\Programs\Python\Python312\python.exe" where UserName is replaced with your user name and use this in orca slicer

krickatthedisco, 2025-01-23:
> This is due to some windows weirdness and I also had this issue. The solution for me was to open PC Settings - Apps - and click on "App execution aliases" and turn them off for the python entries I [...]

I was experiencing the error code 9009, then did the above, and it swapped to error code 2. Unsure of what else I could try to be helpful

Veyraxis, 2025-01-23:
I encountered this same error (9009), tried the above fix. Now it swapped to error code 1.

Ati-93, 2025-01-24:
I tried the Workflow above and as like [krickatthedisco], the error changed to code 2

De-Keizer, 2025-01-24:
Solved the error you need to change -layerheight to -layerHeight then problem is solved

krickatthedisco, 2025-01-24:
> Solved the error you need to change -layerheight to -layerHeight then problem is solved

This was not the issue for me I had the case as intended

rodrykthingol, 2025-01-24:
On linux i also encountered something similar, sometimes it has issues with finding the location of the bricklayers file. 
On Prusa in one computer i would have to /home/user/Folder/bricklayers.py.
On another pc with prusa it would be: /Folder/bricklayers.py
For some reason it was searching for  /home/user/home/user/Folder/bricklayers.py

So try changing the location or omitting the user location on the search file.

DBT85, 2025-01-25:
I have just uninstalled the two Python-related things on my windows machine, installed the latest version and ticked the path button. No change to my error messages after updating them to the new path.

I tried moving the bricklayers.py file to the root of my D drive rather than in a folder with spaces just in case that was causing it but no.

JOMOGO, 2025-01-25:
Slicer: prusa
Python version: 3.10

Code

"C:\Users\user\AppData\Local\Microsoft\WindowsApps\python.exe" "C:\Program Files\Prusa3D\PrusaSlicer\Post processing scripts\bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.3;

C:\Users\user\AppData\Local\Temp\.26744_0.gcode.pp failed.
Error code: 9009

Avantar75, 2025-01-27:
> I get this when I use the "\AppData\Local\Microsoft\WindowsApps\python.exe" path.
> 
> If I use the "\AppData\Local\Programs\Python\Python312\python.exe" path I actually get
> 
> Post-processing script "C:\Users\yourmother\AppData\Local\Programs\Python\Python312\python.exe" "‪D:\Sync\3D Print Archive\bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.1 on file 

This together with a removal and afresh install of Python (installation as administrator and modifying the path both ticked on) made it work for me properly!

krickatthedisco, 2025-01-27:
I updated to the most recent nightly orca and it's working for me now, somewhat. On a multipart object it flattened one of the parts into the correct number of layers but only one layer thick in height.

elfarfadet, 2025-02-10:
Hi,

I manage to get it work (I have 0 knowledge on python or wathever) after I : 

- change -layerheight to -layerHeight
- change the path to AppData\Local\Programs\Python\Python313\python.exe instead of \AppData\Local\Microsoft\WindowsApps\python.exe
- Reboot my computer (I haven't done it since I installed python)

So here an exemple of my post-processing scripts : 

"C:\Users\****\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\*****\Downloads\Bricklayers-main\bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.1;

I'm using Orca 2.2 on Windows 11

Good luck