TengerTechnologies/Bricklayers #3: Not working with Orca Slicer 2.2.0

matejaputic, 2025-01-22:
macOS Sonoma 14.7.2
Orca Slicer 2.2.0

I tried to recreate your example from the video as best as I could, but the script doesn't appear to be working for my setup.

First, in a new project, I created a cube primitive, then set layer height to 0.20 mm, 4 wall loops, 1 top shell layer, 1 bottom shell layer, but 0% infill density.

I tried a few things:

1. Tried adding the Python script to "Post-processing Scripts" section of "Others" in the slicer, as shown in the video. Slicing runs without error. Next, I exported the g-code and opened it in PrusaSlicer G-Code Viewer 2.8.0, but I'm not seeing the offset layers.
2. Tried running the script on the gcode manually, opened result in G-Code Viewer, but same behavior.
3. Tried running a diff between the original gcode file and the modified file, but there are no differences.
4. Tried increasing the debug level to DEBUG and running again, but it seems that it's not detecting the perimeters correctly.

Attached the gcode and the log output.

[brick_cube.gcode.txt](https://github.com/user-attachments/files/18512789/brick_cube.gcode.txt)

[z_shift_log.txt](https://github.com/user-attachments/files/18512788/z_shift_log.txt)

TengerTechnologies, 2025-01-22:
Looks like you have a Bambu printer right? It doesn't work for Bambu printer rn, but I'm working on it.

rjmayott, 2025-01-23:
Happy to help test and diag, I have a few Bambu printers myself

drkpxl, 2025-01-23:
@matejaputic @rjmayott If you can't wait for his code this will work for Bambu printers only not Voron, Creality Etc

[https://github.com/drkpxl/Bricklayers](https://github.com/drkpxl/Bricklayers)

[image]

rjmayott, 2025-01-23:
Very nice! Will test it out now

ProtoBotic, 2025-01-23:
> Looks like you have a Bambu printer right? It doesn't work for Bambu printers rn, but I'm working on it.

It doesnt work for K1 or ender-3 either, the only profile i tried that works is PrusaMini

Dattebayo681, 2025-01-23:
Can confirm, no error code or anything, brick layers just don't form in the exported Gcode when using Orca or Creality K1C

mindforger, 2025-01-23:
i have a sovol sv08 and using orca i also do not get any visible layerchanges

i usually use concentric bottom and surface an monotonic for infill but i dont even see any layer changes on the (3) walls either

todstelzer, 2025-01-23:
Got it working for Bambulab Printers. Try 6 walls.

`# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program. If not, see <https://www.gnu.org/licenses/>.
#
# Copyright (c) [2025] [Roman Tenger]
import re
import sys
import logging
import os
import argparse

# Get the directory where the script is located
script_dir = os.path.dirname(os.path.abspath(__file__))

# Configure logging to save in the script's directory
log_file_path = os.path.join(script_dir, "z_shift_log.txt")
logging.basicConfig(
    filename=log_file_path,
    filemode="w",
    level=logging.INFO,
    format="%(asctime)s - %(message)s"
)

def process_gcode(input_file, layer_height, extrusion_multiplier):
    current_layer = 0
    current_z = 0.0
    perimeter_type = None
    perimeter_block_count = 0
    inside_perimeter_block = False
    z_shift = layer_height * 0.5
    logging.info("Starting G-code processing")
    logging.info(f"Input file: {input_file}")
    logging.info(f"Z-shift: {z_shift} mm, Layer height: {layer_height} mm")

    # Read the input G-code
    with open(input_file, 'r') as infile:
        lines = infile.readlines()

    # Identify the total number of layers by looking for `G1 Z` commands
    total_layers = sum(1 for line in lines if line.startswith("G1 Z"))

    # Process the G-code
    modified_lines = []
    for line in lines:
        # Detect layer changes
        if line.startswith("G1 Z"):
            z_match = re.search(r'Z([-\d.]+)', line)
            if z_match:
                current_z = float(z_match.group(1))
                current_layer = int(current_z / layer_height)

                perimeter_block_count = 0  # Reset block counter for new layer
                logging.info(f"Layer {current_layer} detected at Z={current_z:.3f}")
            modified_lines.append(line)
            continue

        # Detect perimeter types from PrusaSlicer comments
        if ";TYPE:External perimeter" in line or ";TYPE:Outer wall" in line or "; FEATURE: Outer wall" in line:
            perimeter_type = "external"
[... 51 lines cut ...]
            inside_perimeter_block = False

        modified_lines.append(line)

    # Overwrite the input file with the modified G-code
    with open(input_file, 'w') as outfile:
        outfile.writelines(modified_lines)

    logging.info("G-code processing completed")
    logging.info(f"Log file saved at {log_file_path}")

# Main execution
if __name__ == "__main__":
    parser = argparse.ArgumentParser(description="Post-process G-code for Z-shifting and extrusion adjustments.")
    parser.add_argument("input_file", help="Path to the input G-code file")
    parser.add_argument("-layerHeight", type=float, default=0.2, help="Layer height in mm (default: 0.2mm)")
    parser.add_argument("-extrusionMultiplier", type=float, default=1, help="Extrusion multiplier for first layer (default: 1.5x)")
    args = parser.parse_args()

    process_gcode(
        input_file=args.input_file,
        layer_height=args.layerHeight,
        extrusion_multiplier=args.extrusionMultiplier,
    )
`

todstelzer, 2025-01-23:
This works with Bambulab printers. Works with 6 walls too.

https://github.com/todstelzer/Bricklayers-Bambulab

todstelzer, 2025-01-24:
Its working for me. With a Voron 2.4 and a X1C and Outer/Inner wall order.

Do you use a Bambu or a Prusa? For Prusa you can use the original .py as well but you must set the slicer gcode to generate .gcode files. Some people mentioned that its outputing .bgcode by default. Not sure what the issue could be other than that.

todstelzer, 2025-01-24:
Yeah sorry. Heres my post script in Orca maybe its helping
"C:\Users\ddrulez\AppData\Local\Programs\Python\Python313\python.exe" "C:\Brick\bricklayersV2.py" -layerHeight 0.2 -extrusionMultiplier 1.5;

[image]

todstelzer, 2025-01-24:
> Im using a bambu p1s with orcaslicer. I can't find this bgcode setting anywhere but thank you for the help.

Pocke

closed 2025-02-12 by matejaputic