TengerTechnologies/Bricklayers #13: Orca Slicer Error code: 1

BSies0803, 2025-01-23:
Post-processing script "C:\Users\xxx\AppData\Local\Programs\Python\Python313\python.exe"

"C:\Users\xxx\Documents\Bricklayers-main" -layerheight 0.24 -extrusionMultiplier 1.1 on file

C:\Users\xxx\AppData\Local\Temp/orcaslicer_model/Wed_Jan_22/18_29_30#60780#7/Metadata\

.60780.0.gcode.pp failed.

Error code: 1

ParkyGit, 2025-01-23:
I get the exact same issue on PrusaSlicer.

MRuy, 2025-01-23:
I think u missed the file at the end of the path `bricklayers.py`  
Try:  
`"C:\Users\xxx\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\xxx\Documents\Bricklayers-main\bricklayers.py" -layerheight 0.24 -extrusionMultiplier 1.1`

LykleSchepers, 2025-01-23:
No, I have the same error in OrcaSlicer 2.2

Post-processing script "C:\Users\xxx\AppData\Local\Programs\Python\Python38-32\python.exe"
"C:\Program Files\OrcaSlicer\resources\bricklayers.py" -layerHeight 0.28 -extrusionMultiplier 1.1
on file
C:\Users\xxx\AppData\Local\Temp/orcaslicer_model/Thu_Jan_23/15_17_09#21496#49/Metadata\.
21496.0.gcode.pp failed.
Error code: 1

erffrfez, 2025-01-23:
this is only from viewing these comments, I have not read the code, or installed this to try it myself, but back slash vs forward slash: 
...../Metadata
is different from all the rest of that path and in both the posted comments

ParkyGit, 2025-01-23:
I managed to fix it in PrusaSlicer at least. I turned off "Generate BGcode" so that it would make Gcode instead of BGcode.

ParkyGit, 2025-01-23:
I viewed the final gcode in prusaslicer gcode viewer and confirmed the bricklayers worked as well

LykleSchepers, 2025-01-24:
> this is only from viewing these comments, I have not read the code, or installed this to try it myself, but back slash vs forward slash: ...../Metadata is different from all the rest of that path and in both the posted comments

That is in the output of the script. It is in the error msg that is generated.

Kingkuul, 2025-01-24:
> I managed to fix it in PrusaSlicer at least. I turned off "Generate BGcode" so that it would make Gcode instead of BGcode.

i just tried that and now in prusa will actually let me export the file without the error 1  code but when i check it in the gcode viewer it doesnt look like it did anything when comparing to his video  how it looked layer by layer mine just  looked like a normal gcoed

hana34290, 2025-01-25:
In my case, I modified it as follows and it worked!
I partially modified the .py file.

`# Read the input G-code`
`with open(input_file, 'r', encoding='utf-8') as infile:`
`    lines = infile.readlines())`

`# Overwrite the input file with the modified G-code`
`with open(input_file, 'w', encoding='utf-8') as outfile:`
`    outfile.writelines(modified_lines`

voltrare, 2025-01-26:
it works for me with @hana34290 version

kuc-pl, 2025-01-26:
tried to use the script in orca flashforge v1.3.1, sliced view doesn't have bricklayers, attempt to export .gcode file ends with error 1, tried using @hana34290 file mod, same result as well

jubuttib, 2025-01-27:
> it works for me with [[image]@hana34290](https://github.com/hana34290) version on orcaslicer 2.2

How exactly does one implement the fix? I can't figure out where to modify based on those two snippets.

voltrare, 2025-01-27:
> > it works for me with [[image]@hana34290](https://github.com/hana34290) version on orcaslicer 2.2
> 
> How exactly does one implement the fix? I can't figure out where to modify based on those two snippets.

search for those two section ( Read input, Overwrite) and replace

jubuttib, 2025-01-27:
> > > it works for me with [[image]@hana34290](https://github.com/hana34290) version on orcaslicer 2.2
> > 
> > 
> > How exactly does one implement the fix? I can't figure out where to modify based on those two snippets.
> 
> search for those two section ( Read input, Overwrite) and replace

Right, so the only changes I see are the addition of the ", encoding='utf-8'" into the "open(input_file" bits, and an extra ) in "lines = infile.readlines())", neither of which helped. They did slightly change the error code* (still error code 1 though). The formatting is so vastly different in any text editor I have that I can't just directly copy-paste either. Guess I need to wait for a more proper fix.

Oh, I guess there's also a removed ) in "outfile.writelines(modified_lines", but that didn't do anything either...

*Getting either 43064.1.gcode.pp, 47872.0.gcode.pp or 25460.1.gcode.pp depending on which combination of those added and removed ")" I put in.

vavush, 2025-02-02:
> In my case, I modified it as follows and it worked! I partially modified the .py file.
>
> `# Read the input G-code` `with open(input_file, 'r', encoding='utf-8') as infile:` ` lines = [...]

This worked for me, but not immediately. It turned out that there were two separate issues, which I suspect might also be the case for @jubuttib. Here’s what I did:

1. Modify the .py file
Following @hana34290's instructions, I searched for # Read the input G-code and # Overwrite the input file with the modified G-code. I also corrected a misplaced parenthesis in the code:

```
# Read the input G-code
with open(input_file, 'r', encoding='utf-8') as infile:
    lines = infile.readlines()

# Overwrite the input file with the modified G-code
with open(input_file, 'w', encoding='utf-8') as outfile:
    outfile.writelines(modified_lines)

```

2. Fix a case sensitivity issue
Even after modifying the script, I still encountered the same error. Upon closer inspection, I realized that -layerheight should actually be -layerHeight the argument is case-sensitive.

The corrected post-processing script:

`
"C:\Users\here-to-save-the-day\AppData\Local\Programs\Python\PythonXX\python.exe" "C:\Users\here-to-save-the-day\Bricklayers\bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.2;
`

Note: At this point, you may not see any errors, but the changes might not be immediately visible in the Preview. However, if you export the G-code and open it in PrusaSlicer, you should see that the modifications were applied successfully.

Hope this helps! 🚀

jubuttib, 2025-02-02:
@vavush Cheers for trying to help, but even with that I'm not getting anything.

[image]

Dagger101x, 2025-03-21:
Any further successes so far?
neither prusa nor orca work for me.
not even with the mods mentioned above...

thanks in advance. ✌️