TengerTechnologies/Bricklayers #14: UnicodeDecodeError: ‘utf-8’ codec can’t decode byte 0xa1 in position 87: invalid start byte

Olafejs, 2025-01-23:
MacOS + Prusaslicer after generate i have this error:

Post-processing script /usr/bin/python3 “/Users/XXX/Documents/XXX/XXX/bricklayers.py” -layerHeight 0.25 -extrusionMultiplier 1.1 on file /var/folders/fn/_Oyvdbs97rg84l_y9wfh1lqc0000gn/T/.24603_0.gcode.pp failed.
Error code: 1
Output:
Traceback (most recent call last):
File “/Users/XXX/Documents/XXX/XXX/bricklayers.py”, line 139, in process_gcode
File “/Users/XXX/Documents/XXX/XXX/bricklayers.py”, line 46, in process_gcode
lines = infile.readlines()
File “/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/codecs.py”, line 322, in decode
(result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: ‘utf-8’ codec can’t decode byte 0xa1 in position 87: invalid start byte

Any idea where find error ? :)

Olafejs, 2025-01-23:
Ok, error is from binary gcode of MK4S :)

iitggithub, 2025-01-23:
I got the same error. 

I had to update go to Printers > General and uncheck "Supports Binary G-code" and then save the configuration. The script isn't  designed to work with binary g-code files.