TengerTechnologies/Bricklayers #16: Weird layer separation for models with changing wall thickness

wykasz, 2025-01-23:
I'm not sure if it should look like this.
Can it be a problem with rendering gcode?

[image]
[model.zip](https://github.com/user-attachments/files/18522237/model.zip)

PrusaSlicer 2.9.0.
Official profile for Prusa MK3S+ 0.6mm nozzle.
Pretty much default config with 10 perimeters and layer height set to 0.4mm

Command:
"C:\path_to_python\python.exe" "C:\cpath_to\bricklayers.py" -layerHeight 0.4 -extrusionMultiplier 1.1;

With layer height sewt to 0.2mm i had similar results.

Later i'll try to print this gcode to verify if it is printable.

rodrykthingol, 2025-01-23:
Yep saw the same happening while slicing your typical benchy. See the outer gaps thread. Though this seems to be the problem in reverse. Try making the first layer 0.4, external perimeters first, z hopping and try increasing the number of perimeters.

wykasz, 2025-01-23:
@rodrykthingol Thank you for your feedback.

Changing the first layer height to 0.4mm and enabling "external perimeters first" resolved this issue for this simple model.

Setting only the first layer height to 0.4mm resulted in similar artifacts, though they occurred slightly less frequently - every 7th layer instead of every 3rd.

Enabling only "external perimeters first" without changing the first layer height also resolved the issue.

However, this issue still requires a solution that allows the model to be used with all the different slicer settings.
Additionally, this solution has not been validated by me for more complex models.

wykasz, 2025-01-23:
Personally, I’m a bit skeptical about whether it’s possible to achieve more than a proof-of-concept script at the G-code level. My intuition tells me that this approach would have a better chance of success if implemented at the slicing stage. 

That said, I believe @TengerTechnologies’ work is incredibly important, as it enables us all to test this idea on real-life models.

todstelzer, 2025-01-24:
Yeah outer / inner seem to fix it. Must be a issue in the z calculation of some sorts.

rodrykthingol, 2025-01-24:
Hi @wykasz. Honestly it has some bugs, but i managed to print of couple of different models successfully, once i found the tweaks that worked for me. Also did a simple strength test, with and without bricklaying and the way it cracked (exploded) was kinda impressive.

wykasz, 2025-01-24:
So far, with this mod, I have printed close to 500g of filament in the form of Gridfinity-like boxes.

The walls seem to be much stronger while maintaining the same thickness. I conducted a dedicated test print to verify this - it is much harder to snap a wall printed with the bricklayer pattern, and when it does break, it tends to snap across multiple layers rather than along a single layer.

In areas where the mod correctly modifies the G-code, it seems to help a lot. However, I am a bit concerned about more complex geometries where changes to the G-code might introduce weak spots. Additionally, I observe a lot more artifacts on printed models. For example, I notice more imperfections (spots) on vertical surfaces, possibly some over-extrusion in the corners, and a bit more stringing. 

Another limitation is that the current implementation does not work for walls with fewer than three perimeters, which is a downside for me. I print a lot of boxes and structures with thin walls. With this mod, I was able to reduce wall thickness in one of my models, likely without a loss in strength. However, I cannot apply it to models that already have thin walls, as those are printed with only two perimeters. Additionally, I cannot print more than one model at once due to the limitations, which is another drawback.

todstelzer, 2025-01-24:
@wykasz Yeah the brick mod is only applied to inner walls. Outer Walls will get printed normal. 
I did made a Version for Outer walls. Seems to work. Should work for Bambu and Prusa printers.

https://github.com/todstelzer/Bricklayers-Bambulab
[https://github.com/todstelzer/Bricklayers-Bambulab/blob/main/bricklayers_outer_wall.py](url)

xboxhacker, 2025-01-31:
I was able to mod the script to help with these gaps.

**### This is the script from Tenger:**
[image]

**### This is with the mods:**
[image]

I will upload the script later, so it can be tested.

CharlieDroid, 2025-05-25:
@xboxhacker May I know if you were able to upload the script?
I currently have the same problem.
[image]

[image]
The two lines above isn't prominent in the image, but in real life you can really notice it. The filament here is wet, so it is quite stringy.