TengerTechnologies/Bricklayers #18: Pass layer height as variable?

Beulzebub, 2025-01-23:
Hello! I've been doing some digging around the internet and haven't yet found an answer to my question.
This PPS is awesome and is working fantastically on all 3 of my machines - absolutely love it, thank you for the work putting this together and sharing it with us!

Wondering if instead of adding -layerHeight 0.x after the path, I can instead pass through whatever variable the slicer stores as the layer height.
This way I could simply change the layer height in Orca, and it would pass the layer height variable to the python script, instead of me having to remember to change it in both places. I will invariably forget to change it at some point in both spots, looking to fool-proof my process =)

Any help is greatly appreciated, thank you so much!

xboxhacker, 2025-01-23:
Here, try this. This is Tengers code I modified to be a stand-alone Python script. It will run outside of the slicer with a GUI. Click browse, open your gcode file, and click process to make brick layers. It will output a new file with "[fileName]_modified.gcode"

Change the .txt to .py

[bricklayersStandAlone.txt](https://github.com/user-attachments/files/18525981/bricklayersStandAlone.txt)

I am working on some other modifications to the code as well.

Beulzebub, 2025-01-23:
That is awesome!
Is there a way I can pull the temporary .gcode file that Orca outputs first?
Here's what I'm trying to do:
1. Slice the file
2. Poke "Print" (uploads gcode to printer)

Having your new code works great standalone. It would be great to simply add it to the post-process script to execute during the send-to-printer phase.

xboxhacker, 2025-01-23:
Don't that that will work. It is a 'post' processor. The slicer already made the file. The only way to do it is: 
slice>process>drop output file into slicer to see how it looks>re-slice for changes

Beulzebub, 2025-01-23:
Drats, would be super cool. Seems like there'd be a way to automate it. i.e. interrupt the "send to printer" phase.
slice file, save temporary gcode. script runs, imports temporary gcode, runs the script, and sends the modified gcode to the printer.
Pardon my naivety, I've only got an introductory knowledge of python. Leaning on y'all's hard work. Very much appreciated.

Nbeknel, 2025-01-24:
> Is there a way I can pull the temporary .gcode file that Orca outputs first?

Technically it's possible if you use environment variables, the official websites of both Slic3r and PrusaSlicer explain how to use them.

You'd need to add the following to the script:

```
import os
...
layer_height = float(os.environ["SLIC3R_LAYER_HEIGHT"])
```