TengerTechnologies/Bricklayers #8: Z hop when retracting seems to be a mandatory parameter in OrcaSlicer

Schroederr, 2025-01-23:
Orca Slicer 2.2.0
Klipper

Script seems to work fine if Z hop when retracting is set to at least 0.1 in OrcaSlicer. 
But if it is set to 0, when checking the preview of the .gcode, the model is flat on Z0.

[z_hop_0.2.gcode.txt](https://github.com/user-attachments/files/18513737/z_hop_0.2.gcode.txt)

[z_hop_0.gcode.txt](https://github.com/user-attachments/files/18513738/z_hop_0.gcode.txt)

[z_shift_log.txt](https://github.com/user-attachments/files/18513743/z_shift_log.txt)

Beulzebub, 2025-01-23:
Can confirm. Was getting very strange gcode files, like first 6 layers printed all at the first 6 layers, then it would jump to the top layer and try to print the whole rest of the model there. Did some digging, found this comment, fixed the problem entirely. Thank you for pointing this out!

For reference: Ender 3 pro, klipper, mainsail, orcaslicer 2.2.0.

rodrykthingol, 2025-01-23:
Yep in orcaslicer the prints looked like a porcupine

KoenVanduffel, 2025-01-24:
I ran the script "as is" and printed wihout any issue on my RatRig Vcore 500 IDEX. The only thing I did was set the fist layer height identical to all other layers (I normally print the first layer at 0.25 and others at 0.2), maybe that did it, have not checked with a 0.25 mm first layer. I also did not review the gcode in the slicer, maybe the slicer simply can't handle it.