TengerTechnologies/Bricklayers #20: Anyone able to run it on Mac OS?

julyz, 2025-01-24:
Get error :

Post-processing script /usr/local/bin/python3 /Applications/bricklayers.py on file /var/folders/ql/xnzz_48s14jc4xbg7jzyf42c0000gn/T/.50090_0.gcode.pp failed.
Error code: 1
Output:
Traceback (most recent call last):
File "/Applications/bricklayers.py", line 139, in 
process_gcode(
File "/Applications/bricklayers.py", line 46, in process_gcode
lines = infile.readlines()
^^^^^^^^^^^^^^^^^^
File "", line 322, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xce in position 87: invalid continuation byte

iitggithub, 2025-01-24:
Yes, disable binary g-code and it should work.

Instructions are in this github issue:

https://github.com/TengerTechnologies/Bricklayers/issues/14

megalog3d, 2025-01-25:
I don't get an error message anymore, but I don't have a bricklayer either. Any Ideas?

rodrykthingol, 2025-01-26:
Don't have a Mac so i can't test this, but if you go TenTech's youtube channel, theres a couple a videos showing examples of how to run some of his scripts in a Mac.