TengerTechnologies/Bricklayers #22: Layer height error in orca slicer 2.2.0

hamzademirr, 2025-01-24:
When I use it as "C:\Your\Path\To\Python\python.exe" "C:\Your\Path\To\Script\bricklayers.py" -layerheight 0.2 -extrusionMultiplier 1.3 as in the manual, it cannot slice and gives error code 2, but when I delete the -layerheight 0.2 parameter, it slices without any problems. I haven't tried printing the sliced ​​gcode yet. There seems to be no problem in the preview. I plan to write the layer height directly into the .py file.

vkurkchi, 2025-01-24:
It seems like the argument names are case sensitive, did you try setting -layerHeight with capital 'H'?

rodrykthingol, 2025-01-24:
Yep he is correct, theres a typo on the main page it should be "path to python" "path to bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.3

hamzademirr, 2025-01-24:
Thanks for your suggestion, I haven't tried it yet and didn't notice the typo. I'll try it as soon as possible.

UnlawfulFalafel44, 2025-01-24:
I have that same problem, but it still gives me that error even with the correct variable name and path.

I tried deleting the -LayerHeight variable, but then it showed an error for -extrusionMultiplier.

[image]

[image]

todstelzer, 2025-01-24:
@UnlawfulFalafel44 the last line dosent look right.

It should look like this:

[image]

UnlawfulFalafel44, 2025-01-24:
That worked. I must've accidentally put that on a new line. It works now. Thank you