TengerTechnologies/Bricklayers #27: Bambu Compatibility and Layer height Detection

w4fflesjr, 2025-01-24:
I was working on trying to make it compatible with Bambu and also detect the layer height automatically. If anyone wants to try it out let me know if works.
```
import re
import sys
import logging
import os
import argparse
import shutil
from typing import List, Dict, Optional

# Configure logging
logging.basicConfig(
    level=logging.INFO,
    format="%(asctime)s - %(levelname)s - %(message)s",
    handlers=[
        logging.FileHandler("bricklayers.log", mode='w'),
        logging.StreamHandler()
    ]
)

def validate_positive_float(value: float) -> float:
    """Validate that a float is positive."""
    if value <= 0:
        raise argparse.ArgumentTypeError(f"{value} must be positive")
    return value

def process_gcode(input_file: str, extrusion_multiplier: float = 1.0) -> None:
    """
    Main processing function with Bambu compatibility and adaptive layer support
    """
    try:
        # Create backup
        backup_file = input_file + ".bak"
        shutil.copyfile(input_file, backup_file)
        logging.info(f"Created backup at {backup_file}")

        with open(input_file, 'r') as f:
            lines = f.readlines()

        # Detect printer type
        is_bambu = any("Bambu" in line for line in lines[:20])
        pressure_advance = extract_bambu_pressure_advance(lines) if is_bambu else None

        # Analyze G-code structure
        layer_data = analyze_layer_structure(lines, is_bambu)
        modified_lines = process_lines(
            lines, layer_data, extrusion_multiplier, is_bambu, pressure_advance
        )

        # Write modified file
        with open(input_file, 'w') as f:
            f.writelines(modified_lines)
            
        logging.info(f"Successfully modified {input_file}")

    except Exception as e:
        logging.error(f"Error processing file: {str(e)}")
        if os.path.exists(backup_file):
            shutil.move(backup_file, input_file)
            logging.info("Restored original file from backup")
        sys.exit(1)

def analyze_layer_structure(lines: List[str], is_bambu: bool) -> List[Dict]:
    """
    Analyze G-code layers and detect heights
    Returns list of layer dictionaries with:
    - start_line: Line number where layer starts
    - z_height: Absolute Z position
    - height: Layer height
    - is_model: Whether layer contains model (vs support)
    """
    layers = []
    current_z = 0.0
    previous_z = 0.0
    layer_pattern = re.compile(r'^\s*;(LAYER_CHANGE|LAYER:|Bambu layer)', re.IGNORECASE)

    for idx, line in enumerate(lines):
        # Detect layer changes
        if layer_pattern.match(line.strip()):
            layer_height = current_z - previous_z if previous_z else current_z
            layers.append({
                'start_line': idx,
                'z_height': current_z,
                'height': layer_height,
                'is_model': True
            })
            previous_z = current_z

        # Update Z position
        if line.startswith('G1 ') and 'Z' in line:
            if (z_match := re.search(r'Z([\d.]+)', line)):
                current_z = float(z_match.group(1))

        # Detect support material (Bambu)
        if is_bambu and '; support_material' in line.lower() and layers:
            layers[-1]['is_model'] = False

    # Calculate final layer height
    if len(layers) > 1:
        layers[-1]['height'] = current_z - layers[-2]['z_height']
        
    return layers

def process_lines(lines: List[str], layer_data: List[Dict], 
                 extrusion_multiplier: float, is_bambu: bool,
                 pressure_advance: Optional[float]) -> List[str]:
    """
    Process G-code lines with Bambu-compatible modifications
    """
    modified = []
    current_layer_idx = -1
    perimeter_count = 0
    in_perimeter = False

    for idx, line in enumerate(lines):
        # Update current layer
        if any(layer['start_line'] == idx for layer in layer_data):
            current_layer_idx += 1
            current_layer = layer_data[current_layer_idx]
            logging.debug(f"Processing layer {current_layer_idx}")

        # Skip non-model layers
        if not current_layer.get('is_model', True):
            modified.append(line)
            continue

        # Handle Bambu pressure advance
        if is_bambu and 'M900 K' in line:
            modified.append(f"; Original pressure advance: {line.strip()}\n")
            if pressure_advance is not None:
                modified.append(f"M900 K{pressure_advance:.3f} ; Adjusted\n")
            continue

        # Detect perimeter start
        if is_perimeter_start(line, is_bambu):
            in_perimeter = True
            perimeter_count += 1
            modified.extend(handle_z_shift(current_layer, perimeter_count))

        # Adjust extrusion
        if in_perimeter and 'E' in line:
            line = adjust_extrusion(
                line, current_layer['height'], 
                extrusion_multiplier, is_bambu
            )

        # Detect perimeter end
        if in_perimeter and is_perimeter_end(line):
            in_perimeter = False

        modified.append(line)

    return modified

def is_perimeter_start(line: str, is_bambu: bool) -> bool:
    """Detect perimeter start based on printer type"""
    if is_bambu:
        return any(tag in line.lower() for tag in ['outer perimeter', 'inner perimeter'])
    return ';TYPE:Perimeter' in line or ';TYPE:Inner wall' in line

def is_perimeter_end(line: str) -> bool:
    """Detect end of perimeter block"""
    return ';TYPE:' in line or 'G1 F' in line

def handle_z_shift(layer: Dict, count: int) -> List[str]:
    """Generate Z-shift commands with Bambu-compatible feedrates"""
    z_shift = layer['z_height'] + (layer['height'] * 0.5)
    return [
        f"G1 Z{z_shift:.3f} F1200 ; Shifted layer {count}\n",
        f"; Original Z: {layer['z_height']:.3f}\n"
    ]

def adjust_extrusion(line: str, layer_height: float, 
                    multiplier: float, is_bambu: bool) -> str:
    """Adjust extrusion values with height-based compensation"""
    if (e_match := re.search(r'E([\d.]+)', line)):
        base_e = float(e_match.group(1))
        height_factor = layer_height / 0.2  # Reference 0.2mm
        adjusted_e = base_e * multiplier * height_factor
        
        # Preserve existing comments
        comment = line.split(';', 1)[1] if ';' in line else ''
        comment = f" ; Adjusted x{multiplier:.2f} {comment}"
        
        return re.sub(r'E[\d.]+', f'E{adjusted_e:.5f}', line.split(';')[0]) + comment
    return line

def extract_bambu_pressure_advance(lines: List[str]) -> float:
    """Extract original Bambu pressure advance value"""
    for line in lines:
        if 'M900 K' in line:
            if (k_match := re.search(r'K([\d.]+)', line)):
                return float(k_match.group(1))
    return 0.02  # Default Bambu value

def main():
    """Command line interface"""
    parser = argparse.ArgumentParser(
        description="Bricklayer G-code post-processor\nVideo guide: https://youtu.be/EqRdQOoK5hc",
        formatter_class=argparse.RawTextHelpFormatter
    )
    parser.add_argument("input_file", help="Path to input G-code file")
    parser.add_argument("-em", "--extrusion-multiplier", type=float, default=1.0,
                      help="Extrusion multiplier (default: 1.0)")
    
    args = parser.parse_args()
    
    try:
        process_gcode(
            input_file=args.input_file,
            extrusion_multiplier=args.extrusion_multiplier
        )
    except Exception as e:
        logging.error(f"Processing failed: {str(e)}")
        sys.exit(1)

if __name__ == "__main__":
    main()

```

w4fflesjr, 2025-01-24:
python bricklayers.py input.gcode -em 1.2

bhast2, 2025-01-25:
> python bricklayers.py input.gcode -em 1.2

```
Post-processing script "C:\Users\xxxxxx\AppData\Local\Microsoft\WindowsApps\python.exe"
"C:\Users\xxxxx\Desktop\bricklayers.py" on file
C:\Users\xxxxxxx\AppData\Local\Temp/orcaslicer_model/Sat_Jan_25/08_59_40#3848#55/Metadat
a\.3848.0.gcode failed.
Error code: 9009
```
This is the error I am getting with yours not sure if I am doing it right. 

This is what I have in my pps section.

`"C:\Users\xxxx\AppData\Local\Microsoft\WindowsApps\python.exe" "C:\Users\xxxx\Desktop\bricklayers.py"`

w4fflesjr, 2025-01-25:
OK here's an updated code that I've gotten to print with successfully using OrcaSlicer with an Ender 3.
Post Processing Script is ```"C:\Python313\python.exe" "C:\Users\YOURPATH\bricklayers.py" -em 1.3;```
```import re
import sys
import logging
import os
import argparse
import shutil
from typing import List, Dict

# Configure logging
logging.basicConfig(
    level=logging.INFO,
    format="%(asctime)s - %(levelname)s - %(message)s",
    handlers=[
        logging.FileHandler(os.path.expanduser("~/bricklayers.log"), mode='w'),
        logging.StreamHandler()
    ]
)

def process_gcode(input_file: str, extrusion_multiplier: float = 1.0) -> None:
    """Process G-code in place for OrcaSlicer compatibility"""
    try:
        # Validate input file exists
        if not os.path.isfile(input_file):
            raise FileNotFoundError(f"Input file not found: {input_file}")

        logging.info(f"Processing file: {input_file}")

        # Read original content
        with open(input_file, 'r') as f:
            original_lines = f.readlines()

        # Create backup in home directory
        backup_dir = os.path.expanduser("~/bricklayer_backups")
        os.makedirs(backup_dir, exist_ok=True)
        backup_path = os.path.join(backup_dir, os.path.basename(input_file))
        shutil.copyfile(input_file, backup_path)
        logging.info(f"Backup created at: {backup_path}")

        # Process lines
        modified_lines = process_lines(original_lines, extrusion_multiplier)

        # Write back to original file
        with open(input_file, 'w') as f:
            f.writelines(modified_lines)
        logging.info("File successfully modified in place")

    except Exception as e:
        logging.error(f"Error processing file: {str(e)}", exc_info=True)
        raise

def process_lines(lines: List[str], extrusion_multiplier: float) -> List[str]:
    """Process G-code lines without filename changes"""
    output = []
    current_z = 0.0
    layer_height = 0.2  # Will be calculated dynamically
    in_perimeter = False
    perimeter_count = 0

    for line in lines:
        # Track layer changes
        if line.startswith(';LAYER_CHANGE'):
            if 'Z=' in line:
                new_z = float(re.search(r'Z=([\d.]+)', line).group(1))
                layer_height = new_z - current_z
                current_z = new_z
                perimeter_count = 0
                logging.debug(f"New layer at Z{current_z}, height: {layer_height}")

        # Detect perimeter starts (OrcaSlicer pattern)
        if ';WALL-OUTER' in line or ';WALL-INNER' in line:
            perimeter_count += 1
            in_perimeter = True
            if perimeter_count % 2 == 1:
                shifted_z = current_z + (layer_height * 0.5)
                output.append(f"\n;BRICKLAYER MOD START\nG1 Z{shifted_z:.3f}\n")
        
        # Modify extrusion in perimeters
        if in_perimeter and ' E' in line:
            line = adjust_extrusion(line, extrusion_multiplier, layer_height)

        # Detect perimeter ends
        if in_perimeter and ';TYPE' in line:
            in_perimeter = False
            if perimeter_count % 2 == 1:
                output.append(f"G1 Z{current_z:.3f}\n;BRICKLAYER MOD END\n")

        output.append(line)

    return output

def adjust_extrusion(line: str, multiplier: float, layer_height: float) -> str:
    """Adjust extrusion values proportionally to layer height"""
    e_match = re.search(r'E([\d.]+)', line)
    if not e_match:
        return line
    
    original_e = float(e_match.group(1))
    adjusted_e = original_e * multiplier * (layer_height / 0.2)
    
    return re.sub(
        r'E[\d.]+', 
        f'E{adjusted_e:.5f}', 
        line
    ) + f" ; Adjusted (x{multiplier:.2f})"

def main():
    parser = argparse.ArgumentParser(
        description="OrcaSlicer-compatible in-place G-code modifier",
        add_help=False
    )
    parser.add_argument("input_file")
    parser.add_argument("-em", "--extrusion-multiplier", type=float, default=1.0)
    
    args = parser.parse_args()
    
    try:
        process_gcode(
            args.input_file,
            args.extrusion_multiplier
        )
    except Exception as e:
        logging.error(f"Fatal error: {str(e)}")
        sys.exit(1)

if __name__ == "__main__":
    main()```

w4fflesjr, 2025-01-25:
i'm have issues with artifacts in the print do not try and use this

closed 2025-01-25 by w4fflesjr