TengerTechnologies/Bricklayers #29: MCU 'mcu' shutdown: Rescheduled timer in the past

cons30rus, 2025-01-24:
I tried the script and faced to printer suddenly stopped after couple hours of work. Klipper log says:

[2025-01-24 16:46:19,437][INFO]:MCU 'mcu' shutdown: Rescheduled timer in the past
clocksync state: mcu_freq=72000000 last_clock=29117771830872 clock_est=(404391.900 29115704577862 71998886.682) min_half_rtt=0.000193 min_rtt_time=402139.809 time_avg=404391.900(856.787) clock_avg=29115704577862.855(61687710132.808) pred_variance=23060357.631
Dumping serial stats: bytes_write=1344765560 bytes_read=214886616 bytes_retransmit=28772 bytes_invalid=0 send_seq=23337372 receive_seq=23337369 retransmit_seq=23337253 srtt=0.003 rttvar=0.001 rto=0.025 ready_bytes=73 stalled_bytes=2967

I never had this error before with regular printing

cons30rus, 2025-01-24:
According klipper documentation this error indicates an overloaded / jammed host, this error points to an overloaded MCU.
https://klipper.discourse.group/t/rescheduled-timer-in-the-past/6635
The printer uses  stm32f103xe with CLOCK_FREQ=72000000 MCU. Klipper documenation says this MCU has 3 stepper = 264
In dump log interval=75. It is low than 264 and causes the error.

cons30rus, 2025-01-25:
I think slicer made gcode file with maximum printer resource utilization and additional movement from script overload it.

rodrykthingol, 2025-01-25:
Weird. Thats a pretty modern printer. I used to have those errors with the old 8 bit motherboards. Honestly before messing with the config file, check the cable that connects the host running klipper to the main motherboard. Most of the times its just a crappy usb connection. If it doesn't work, upgrade the printer firmware, the flsun s1 is a new printer and we all know that chinese don't really do quality control or beta testing a product. Also an old trick was using arc fitting to reduce the motherboard usage.