TengerTechnologies/Bricklayers #31: Complex Geometry causes Surface defects

theboee, 2025-01-25:
Complex geometry causes surface defects to appear.

[image]

May cause holes, missing layers, print failures, tolerance issues, etc....

Also, offset layers do not appear to work on large overhangs. It acts like a standard sliced part, until the next layer where it continues with the original offset. This will undoubtedly create holes or large gaps between layers.

If this scenario is too complex to fix in python, have the code apply the final correction layer to the layer with the large overhang and then start the offsetting again. (Act like the layer with the overhang is the final layer to even everything out and apply the offset layers 1-2 layers after it.)

As this code stands right now it only seems useful for very simple geometry. Hope to see an update that fixes it.

rodrykthingol, 2025-01-25:
Hi @theboee could you please show me the gcode and the log file? Also try slicing with first layer = layer height, external perimeters first, z hop activated and try increasing the number of perimeters if all else fails. That seems to be the layer gap bug.

theboee, 2025-01-27:
Changing to outer/inner makes it appear that the surface defects are gone but all it does is move the defects to the inside of the part instead. Also there appears to be something going on with line widths.

theboee, 2025-01-27:
cant get any files to upload. would send them if i could