TengerTechnologies/Bricklayers #32: Complex model shows up strange in gcode viever

atomek321, 2025-01-25:
[z_shift_log.txt](https://github.com/user-attachments/files/18545475/z_shift_log.txt)

[Arm_ABS_45ms.gcode.txt](https://github.com/user-attachments/files/18545479/Arm_ABS_45ms.gcode.txt) remove .txt to open 

is it normal for it to show up like that in viever? it looks like it goes back to Z 0.25 every few lines

rodrykthingol, 2025-01-25:
Hi @atomek321 Haven't seen a bug like that before, for some reason the app only counted one layer. If you look at the log it keeps refering to layer 1 over and over. Thats why you keep seeing the g1 z0.250 command. The most similar bug i've seen is a bug where layer 1 and 2 had also the same z height, it caused a elephants foot, but it counted the layers correctly.

Orcaslicer been giving some weird bugs, i read that someone was capable of solving some of the  problems by disabling binary gcodes, i believe the post was made in one of the opened orcaslicer issues threads. Will have to check. Also try gcode flavor Marlin or Klipper.

On prusa slicer i can confirm that you can slice with scarf seams. Best practices are first layer same as layer height, external perimeters first, z hop activated and sometimes you may need to increase the number of perimeters.

But you should always check the gcode produced. Most commons bugs are perimeters gaps and elephants foot due to layer stacking, though in your case, seems like the app failed to calculate the number of layers on the part.

atomek321, 2025-01-25:
im usin voron trident with klipper.
i managed to isiolate this isue with strange layers to when i DISABLE Scarf seam.
With Scarf seam on it creates more managable gcode. witch i can print but mu printer detects clog. so something is with layers stacking anyway. 

test files with enabled and disabled scarf seam

[z_shift_log enabled.txt](https://github.com/user-attachments/files/18547886/z_shift_log.enabled.txt)

[WheelHalf Enabled.gcode.txt](https://github.com/user-attachments/files/18547887/WheelHalf.Enabled.gcode.txt)

[z_shift_log scarf disabled.txt](https://github.com/user-attachments/files/18547889/z_shift_log.scarf.disabled.txt)

[WheelHalf test scarf disabled.gcode.txt](https://github.com/user-attachments/files/18547891/WheelHalf.test.scarf.disabled.gcode.txt)

rodrykthingol, 2025-01-26:
@atomek321 

o.O Thats... Could you please send me the stl's so i could mess around a bit?