TengerTechnologies/Bricklayers #33: Post Processing script error: Error code: 9009

GijsGroote, 2025-01-25:
What is the cause of this error?

Post-processing script "C:\Users\gijsg\AppData\Local\Microsoft\WindowsApps\python.exe" "C:\Users\gijsg\Documents\bricklayers.py" on file C:\Users\gijsg\AppData\Local\Temp\.14736_0.gcode.pp failed.
Error code: 9009

[image]

check: binary mode is off
check layer height matches the specified layer height

layer height 0.2 mm

post processing script in settings is set to:

"C:\Users\gijsg\AppData\Local\Microsoft\WindowsApps\python.exe" "C:\Users\gijsg\Documents\bricklayers.py"

And 

"C:\Users\gijsg\AppData\Local\Microsoft\WindowsApps\python.exe" "C:\Users\gijsg\Documents\bricklayers.py"-layerheight 0.2 -extrusionMultiplier 1.3

I encountered the same on Linux, I have tried multiple STL files, any idea?

atomek321, 2025-01-25:
reinstal python you need to use both of these options

[image]

iitggithub, 2025-01-25:
It can be worthwhile attaching the z_shift_log.txt to these issues for review as well. It should be found in the same directory that bricklayers.py is in.

atomek321, 2025-01-25:
error does not create them becouse its not finding corect path to python.
script is not executed

wykasz, 2025-01-25:
> reinstal python you need to use both of these options
> 
> [image]

That is a missing step in the instruction and should be added to the manual.

rodrykthingol, 2025-01-25:
Also wanted to point out that theres a typo in the readme its -layerHeight 0.2, not -layerheight 0.2

Please try this:
"C:\Users\gijsg\AppData\Local\Microsoft\WindowsApps\python.exe" "C:\Users\gijsg\Documents\bricklayers.py"-layerHeight 0.2 -extrusionMultiplier 1.3

Also my recommendation for a sucessful print is use prusaslicer, first layer = layer height, marlin/klipper gcode, z hop activated, external perimeters first and sometimes increase the number of perimeters will solve some issues.

iitggithub, 2025-01-26:
> error does not create them becouse its not finding corect path to python. script is not executed

Even though the path to python is hard coded in the script?

`
"C:\Users\gijsg\AppData\Local\Microsoft\WindowsApps\python.exe"
`

GijsGroote, 2025-01-26:
Hmm Error codes have changed, I've reinstalled python3.13 with the options 
* use admin privilegeswhen installing py.exe
* Add python.exe to path
Both checked.

Instead of the error code 9009, I get error codes 1 and 2. 

Post Processing Scripts:
"C:\Users\gijsg\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\gijsg\Documents\bricklayers.py"-layerHeight 0.2 -extrusionMultiplier 1.3

Gives Error:
Post-processing script "C:\Users\gijsg\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\gijsg\Documents\bricklayers.py"-layerHeight 0.2 -extrusionMultiplier 1.3 on file C:\Users\gijsg\AppData\Local\Temp\.16000_0.gcode.pp failed.
Error code: 2

Post Processing Scripts:
"C:\Users\gijsg\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\gijsg\Documents\bricklayers.py"

Gives Error:
Post-processing script "C:\Users\gijsg\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\gijsg\Documents\bricklayers.py" on file C:\Users\gijsg\AppData\Local\Temp\.16000_0.gcode.pp failed.
Error code: 1

I'd like to provide the z_shift_log.txt but it's never created.

bucher87, 2025-01-27:
I had this exact same issue, followed the recommendations here, and long story short, CMD spit back at me was there was an error writing z_shift_log.txt. Moved the script to a non protected folder ("c:\Users\*****\Documents\bricklayers.py") and that fixed the issue. Also made sure the path was "C:\Users\*****\AppData\Local\Programs\Python\Python313\python.exe" and not the Winapps one.

GijsGroote, 2025-01-28:
# Using BrickLayers with G-code Files

I've generated a G-code file (the Benchy boat) with the default settings, called `brick_this_benchy.gcode`, then ran `bricklayers.py` from the CMD (instead of using the output options in PrusaSlicer) as follows:

```bash
python bricklayers.py brick_this_benchy.gcode -layerHeight 0.2 -extrusionMultiplier 1.3
```

The `z_shift_log.txt` indicates no errors, but visual inspection using the PrusaSlicer G-code Viewer shows no bricklayers. Creating the bricklayers actually worked fine, but they are not visible in the PrusaSlicer G-code Viewer. However, when using the BambuLab slicer to view the `brick_this_benchy.gcode` file, the bricklayers are visible.

## Current Workflow

**This is fine for me; it works but is not ideal:**

1. Create the `brick_this_file.gcode` file with PrusaSlicer.
2. Generate bricklayers by running `bricklayers.py` on the `brick_this_file.gcode` file in the CMD as shown above.
3. Check the file `brick_this_file.gcode` using the BambuLab slicer.
4. If `brick_this_file.gcode` does not look as expected, inspect `z_shift_log.txt`.
5. If everything looks fine, print `brick_this_file.gcode`.

## Future Expectations

I expect that bricklayers will become an actual feature in slicers within the coming years. I will only temporarily use this workaround.

The [z_shift_log.txt](https://github.com/user-attachments/files/18577746/z_shift_log.txt) from calling bricklayers.py via the CMD.

iitggithub, 2025-01-29:
> Hmm Error codes have changed, I've reinstalled python3.13 with the options
>
> * use admin privilegeswhen installing py.exe
> * Add python.exe to path
>   Both checked.
>
> Instead of the error code 9009, I [...]

The above error code is likely because you have a typo there. There should be a space in front of the -layerHeight option like so:

`
"C:\Users\gijsg\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\gijsg\Documents\bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.3
`

> Post Processing Scripts: "C:\Users\gijsg\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\gijsg\Documents\bricklayers.py"
>
> Gives Error: Post-processing script [...]

This one *could* be because that directory isn't writable and the z_shift_log.txt can't be created but it's difficult to confirm because there are many functions in python that could return an exit code of 1. Having said that, i assume you're running Prusaslicer as the gijsg user so it *should* be able to write to that directory.

Unfortunately, the only way to confirm the exact error is by running the script manually via the command line. I'm not sure why (maybe it's a windows thing?) but Prusaslicer is only showing you the error code when it should show the error code and output. The output doesn't depend on the z_shift_log.txt file existing.

Once the error is resolved, you should be able to run the script as a post-processing script straight from the slicer.