TengerTechnologies/Bricklayers #34: Add automatic layer height detection from G-code comments

iitggithub, 2025-01-25:
Currently, the layer height needs to be passed to the script via a command line parameter. In prusa slicer, I would need to have the post processing script set to the following on my mac:

`"/Library/Frameworks/Python.framework/Versions/Current/bin/python3" "/Users/stutaki/iitggithub/Bricklayers/bricklayers.py" -layerHeight 0.2;`

The script should use the G-code comments to determine the layer height the model was sliced at.

During testing i found that Prusa printers seem to mention the layer height like so:

`
; layer_height = 0.15
`

and from some digging around online, I believe Bambu Labs uses:

`
; Z_HEIGHT: 0.15
`

iitggithub, 2025-01-25:
Submitted Pull request https://github.com/TengerTechnologies/Bricklayers/pull/26 to add automatic layer height detection via G-code comments

taotien, 2025-01-25:
according to https://help.prusa3d.com/article/post-processing-scripts_283913, you can also base it off of an environment variable. I'm assuming it's gonna be SLIC3R_LAYER_HEIGHT

iitggithub, 2025-01-27:
> According to https://help.prusa3d.com/article/post-processing-scripts_283913, you can also base it off of an environment variable. I'm assuming it's gonna be SLIC3R_LAYER_HEIGHT.
> 
> Although, it might be more convenient to just detect from the file for users running post-processing manually.

I've pushed a commit that will try to obtain the layer height from the SLIC3R_LAYER_HEIGHT environment variable. If that fails, it'll try to find the layer height from the G-code comments and if that fails, it'll use the command line argument -layerHeight. Even if -layerHeight isn't specified it does set a default value of 0.2mm so we should be safe.

AJolly, 2025-02-28:
ideally we update the layerHeight detection as we go along - I know I don't always have the same layer height throughout my entire model.