TengerTechnologies/Bricklayers #38: Error code: 2 Using Orca Slicer

darraghh1, 2025-01-27:
Post-processing Scripts: 
          "C:\Program Files\Python313\python.exe" "C:\Program Files\OrcaSlicer\scripts\bricklayer.py" -layerHeight 0.2

Orca Slicer Error:
  Post-processing script "C:\Program Files\Python313\python.exe" "C:\Program
  Files\OrcaSlicer\scripts\bricklayer.py" -layerHeight 0.2 on file
  C:\Users\darra\AppData\Local\Temp/orcaslicer_model/Sun_Jan_26/19_12_14#4604#49/Metadat
  a\.4604.0.gcode.pp failed.
  Error code: 2

C:\Users\darra>where python
C:\Program Files\Python313\python.exe

todstelzer, 2025-01-27:
If i change the Bricklayer script name i get the same error. Seems like your path to the script could be the issue.

[image]

sergid98, 2025-01-30:
I think the problem is with the H in -layerHeight 0.2.
I had the same error when copying the command from GitHub, so try changing -layerheight 0.2 to -layerHeight 0.2.

cosmocin1, 2025-02-01:
i have same issue is show same error i try changing -layerheight 0.2 to -layerHeight 0.2.

80163916, 2025-02-03:
I have same issue，the slicer software is PrusaSlicer，that version is 2.9.0，Python version is 3.12.8，my computer os is win11 64.
error info：
Post-processing script "C:\Users\weiji\AppData\Local\Programs\Python\Python312\python.exe" "C:\Users\weiji\Documents\3D prient\bricklayers.py" -layerheight 0.2 -extrusionMultiplier 1.3 on file C:\Users\weiji\AppData\Local\Temp\.40828_0.gcode.pp failed.
Error code: 2

[image]

todstelzer, 2025-02-03:
> I have same issue，the slicer software is PrusaSlicer，that version is 2.9.0，Python version is 3.12.8，my computer os is win11 64. error info： Post-processing script [...]

Are you sure your path is correct? 
 "C:\Users\weiji\Documents\3D prient\bricklayers.py  „3D prient?“

DanielEikelenboom, 2025-02-05:
Capitalization is important. The example line on the homepage is incorrect.

Use `"C:\Your\Path\To\Python\python.exe" "C:\Your\Path\To\Script\bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.3` like @sergid98 suggested.