TengerTechnologies/Bricklayers #40: win 32 error 5

shikyojojo, 2025-01-27:
Not sure if I am doing this wrong or not but this is the error I'm getting Failed starting the script C:\Users\Evan\AppData\Local\Programs\Python\Python313 C:\bricklayer\bricklayer.py -layerheight 0.2 -extrusionMultiplier 1 C:\Users\Evan\AppData\Local\Temp\.18164_0.gcode.pp, Win32 error: 5

I made sure to have the run admin box and the other clicked during install of python.

MicroSur, 2025-01-31:
C:\Users\Evan\AppData\Local\Programs\Python\Python313\python.exe

shikyojojo, 2025-02-03:
Tried that and got this 

[image]

Maurice3er, 2025-02-09:
> Tried that and got this
> 
> [image]

there is a typo in the README which you copied over: "-layerheight 0.2". The parameters are case sensitive. Try it out with "-layerHeight 0.2" or even without layerHeight parameter as it should default to 0.2mm.

shikyojojo, 2025-02-12:
without layerheight and still got this, not sure what im doing wrong. 

[image]

xM4URiiC3x, 2025-02-12:
Was there a log file written in the directory of the Python script? If yes, attach it. You might also want to include your input of the 'Post-Processing' tab and not only the error message for analysis of the issue.