TengerTechnologies/Bricklayers #41: Orca Slicer Layer Height Issue

MikalTim, 2025-01-27:
I have been trying to get he script working in orca slicer and Prusa. The issue (if it is indeed the same issue for orca and Prusa) is that the layer heights get messed up (and in the case of orca, almost zeroed). Its only the bottom layer that has the correct height. In Prusa I have managed to fix the issue by setting the layer hight to the nozzle width. As of orca, I have tried many settings with no success. I have set it to relative E distance with no success. Here is an image of a simple cube slice:

[image]

V037, 2025-02-01:
meanwhile there is me who I don't understand if it works for me or not

Protonova, 2025-02-07:
> meanwhile there is me who I don't understand if it works for me or not

Funny, same here. Glad I wasn't the only one...

I got it to output "valid" gcode, but I didn't see any changes. I zoomed in spent time looking at the tool path and couldn't really see a change in layer ordering.

Off topic:
I'm sure @TengerTechnologies is being overwhelmed by this influx of issues/pull requests. Give him some time. Honestly I'm surprised we haven't had someone step up and coordinate pull requests. Essentially review and condense pull requests as he is only one person working only on the weekend if even lol.

Danzaywer, 2025-04-03:
I have the same problem, each layer is considered with height 0.2 and the resulting file always prints at 0.2 and 0.3, the tree supports are not modified and appear normal.

Danzaywer, 2025-04-03:
I found the cause, there are only one "G1 Zx.x" line into the file at the beginning, generate by Orca, other layer changes are combined into one line "G1 Xx.x Yx.x Zx.x", now the question is WHY? I tryed to put "G1 Z[layer_z]" in layer change gcode section and now the shift is done but the first layer is wong due the double "G1 Z" , i thing can solve with an {if layer_num>1} in layer change gcode.

Danzaywer, 2025-04-03:
Ok, I modified the script and now I have a consistent gcode in preview.
the modification I made wos:
line 55: change "if line.startswith("G1 Z"):" into  "if line.startswith("G1") and "Z" in line:"
lines 56,100,106,111,116:  change "r'E[-\d.]+'" into "E[-\d.\d]+'" 
line 80: add "and current_layer > 1 and current_layer < total_layers" to the "if" statement to exclude first and last layer
line 103: change in "if current_layer == 2"
My Orca slicer consider 1 as first layer and total_layers as last, not 0 and total_layers-1.
I have not send  the gcode to printer yet.

[bricklayersMOD.py.txt](https://github.com/user-attachments/files/19590316/bricklayersMOD.py.txt)