TengerTechnologies/Bricklayers #50: Testing - BricklayerNonPlanar -

rodrykthingol, 2025-02-01:
Every odd perimeters gets its z height radically increased.

[image]

[Shape-Box.gcode.txt](https://github.com/user-attachments/files/18629973/Shape-Box.gcode.txt)
[z_shift_log.txt](https://github.com/user-attachments/files/18629974/z_shift_log.txt)

TengerTechnologies, 2025-02-01:
Can you maybe slice the file once without the script and once with the script and attach both gcode files?

TengerTechnologies, 2025-02-01:
Updated the layerheight detection to include ;
Can you test it again with the new version please

rodrykthingol, 2025-02-01:
[image]

Fixed.

rodrykthingol, 2025-02-01:
Fixed g-code and log.
[Shape-Box_fixed.gcode.txt](https://github.com/user-attachments/files/18630033/Shape-Box_fixed.gcode.txt)
[z_shift_log.txt](https://github.com/user-attachments/files/18630034/z_shift_log.txt)

TengerTechnologies, 2025-02-01:
Great!

rodrykthingol, 2025-02-01:
It seems it has some issues with travel moves. Its extruding in some travel moves.

[image]

[3dbenchy_bugged.gcode.txt](https://github.com/user-attachments/files/18630043/3dbenchy_bugged.gcode.txt)
[3dbenchy_vanilla.gcode.txt](https://github.com/user-attachments/files/18630042/3dbenchy_vanilla.gcode.txt)
[z_shift_log.txt](https://github.com/user-attachments/files/18630041/z_shift_log.txt)

rodrykthingol, 2025-02-01:
And this is kinda minor, but theres this issue with the amplitude fade, resulting in a top layer interfering with the sin wave of the infill.

[image]

rodrykthingol, 2025-02-01:
Correction the interfering layer is designated as a bridge.

rodrykthingol, 2025-02-01:
Did some extra testing.

Wall Reorder OFF:

[image]

Wall Reorder ON:

[image]

Wall Reorder ON + External Perimeters OFF

[image]

Wall Reorder ON + External Perimeters OFF + Nearest SEAM (,instead of aligned)

[image]

Seems like you have to avoid travel moves and or only use in models with a natural progression from inner to outer perimeters, maybe contiguous perimeters?

Wall Reorder ON + External Perimeters OFF + Nearest SEAM + Classic (Disabled Arachne)

[image]

Though honestly its a bit cosmetic, there are some travel move inside that might be problematic.

[image]

rodrykthingol, 2025-02-01:
Not much of a programmer, but is it possible to recognize travel moves? Maybe do a simple logic that calculates the distance between two points and if its larger than say wall width? nozzle diameter? Then it disables the extrusion?

Legohead259, 2025-02-02:
Want to contribute another datapoint:

Prusa XLIS, 0.4mm nozzle, 0.2mm layer height, external perimeters first: true
`-extrusionMultiplier 1.05 -wallReorder 0 -nonPlanar 1 -amplitude 0.6 -frequency 1.1`

[image]

rodrykthingol, 2025-02-03:
Huh. Those are some interesting travel lines. And yeah, turning wallReorder off seems to make things even worse.

TekuSP, 2025-02-04:
[image]

[image]

I have interesting results as well :)

krickatthedisco, 2025-02-11:
I've begun testing a few different models (cubes, benchy, and a big gear) with the newest script in orcaslicer dev 2.3 and the bricklayers work great but I cannot get the nonplanar infill to work for the life of me. I've tried the 0/1 enable disable and playing around with the amplitude to no effect, always just regular infill in the gcode preview

CaitsSB, 2025-02-11:
> I've begun testing a few different models (cubes, benchy, and a big gear) with the newest script in orcaslicer dev 2.3 and the bricklayers work great but I cannot get the nonplanar infill to work [...]

Same here, brick layers are working fine apparently, but nonPlanar Infill is nowhere to be found, tried every possible combination as far as I know using both Prusa Slicer 2.9.0 and Orca Slicer 2.2.0