TengerTechnologies/Bricklayers #52: Non Linear Infill/Brick Layers - Odd Behaviour

lechuck333, 2025-02-02:
I just tried the combination script on a random stl file I had on my computer. The behaviour is quite weird. It looks like the infill is leaving the shell of the print.

[image]

I've attached the gcode (renamed as a text file) and the log file.

[MXR Pedal Holder F360_0.4n_0.2mm_PLA_MK4S_1h21m - Copy.txt](https://github.com/user-attachments/files/18630708/MXR.Pedal.Holder.F360_0.4n_0.2mm_PLA_MK4S_1h21m.-.Copy.txt)

[z_shift_log.txt](https://github.com/user-attachments/files/18630709/z_shift_log.txt)

The script settings are: -extrusionMultiplier 1.05 -wallReorder 1 -nonPlanar 1 -amplitude 0.6 -frequency 1.1;

dennyblokland, 2025-02-03:
My two cents: 
First of all, I love the idea of one day using this and I'm really excited you've written the script to get this started. 

Second: I've also gotten weird behavior as a result. (OrcaSlicer 2.2.0, MacOS, PrusaXL)

I'm running Bambu Lab printers and the G-Code after running script didn't seem to indicate any brick layers, so I tried using a Prusa printer profile (PrusaXL is available in  OrcaSlicer so I picked the PrusaXL). 

It now changed the G-Code, but the model has become inverted kind of (going partially below the build plate), and generally ripped apart. The first layer seems alright, the next few layers seem to be collapsed into each other and the rest shows a narrow pillar instead of the (wider) shape going up, and an opposite, different extrusion going below the build plate. 

... Bricklayers/bricklayers.py" -layerHeight 0.2 -extrusionMultiplier 1.1

Another model, just tried, just collapses itself completely in the first 4 (phyiscal) layers, so imagine 1000 layers squished into four. I do seem to recognize some alternating (half) layer heights at the beginning, though. 

I mainly just wanted to share my n=1 datapoint of experience.

 Hope this helps.

rodrykthingol, 2025-02-03:
Yep i've seen that happen. For some reason the script forgets to increment layer height.