TengerTechnologies/Bricklayers #53: No error code, but no changes to the sliced model

Nyamogus, 2025-02-02:
[image]
I haven't tried it on Linux so far, but the script doesn't seem to apply itself to the model

rodrykthingol, 2025-02-02:
Hi @Nyamogus. Haven't managed to make it work in orcaslicer, only prusaslicer. But to run the script you need to export the gcode file.

Kingkuul, 2025-02-02:
im having half of this issue i can get the Non Planar Infill to work but not the brick layering, im on the latest version of pursa 2.9.0

Titanyus, 2025-02-02:
@Nyamogus The script is applied while generating G-Code, NOT while slicing!
What happens if you press "Export G-Code"?

Nyamogus, 2025-02-02:
@Titanyus That actually worked now that i tried it again. When i did that initially i accidentally sliced on my Bambu profile and not the Voron one

closed 2025-02-02 by Nyamogus

mkebaish, 2025-06-18:
> [@Nyamogus](https://github.com/Nyamogus) The script is applied while generating G-Code, NOT while slicing! What happens if you press "Export G-Code"?

UPDATE!!!

i tried to make export to gCode >>> SO i got this massege >>> whats it mean

(((Post-processing script "C:\Users\Mo
Kebaish\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\Mo
Kebaish\AppData\Local\Programs\Python\Python313\Scripts\bricklayers.py" -layerHeight 0.2
extrusionMultiplier 1 on file
C:\Users\MOKEBA~1\AppData\Local\Temp/orcaslicer_model/Wed_Jun_18/05_23_42#14720#55/M
etadata\.14720.0.gcode.pp failed.
Error code: 2)))