TengerTechnologies/Bricklayers #55: Wierd gcode generated.

EmilHaase, 2025-02-04:
I am not sure if the issue is on my side, but nevertheless, it is interesting.

"C:\Users\USER\AppData\Local\Programs\Python\Python313\python.exe" "C:\Users\USER\Downloads\bricklayersNonPlanarInfill.py" -extrusionMultiplier 1.05 -wallReorder 0 -nonPlanar 1 -amplitude 0.6 -frequency 1.1;

External perimeter first.

Set to output as gcode because whan set to bgcode it failed.

Also the non planar infill looks like it generated but alsothe wierd circles.

Thank you for making improvements to this community.

[image]

captainmustard, 2025-02-14:
the circles are arcs. you've either got arc fitting enabled in orca or you're using arc fitter. Nothing to do with bricklayer.

GilesBathgate, 2025-02-21:
Its seems that bricklayersNonPlanarInfill.py is completely broken when using Arc Fitting G2/3IJ setting in prusa slicer.

It would be nice if the script could detect that setting and provide an error, it would also be good to detect binary gcode and display an error too.