TengerTechnologies/Bricklayers #56: [Question] README says that bambu printers aren't supported.

MagicBOTAlex, 2025-02-04:
I'm just wondering if it is bambu printers not working, or if it is just the bambu slicer that is not working.
Because I use orca slicer for my bambu p1s, and I fear that something bad is going to happen if I try this.

LumiereMarine, 2025-02-04:
I believe it is the printers. I have an X1C, and have been trying to use Claude and Gemini AI to code it for the X1C (IDK anything about Python lol), and there seems to be a lot of differences in the wording of how the printer prints. I have been trying to get something to work on the bambu, but it keeps saying invalid gcode, so there's still some issues. I'm wondering if there's ultimately something bambu uses to detect that needs to be added? Not much info I've been able to find on post processing scripts for it. I've managed to add the post processing script in and export gcode, but after running the script in python, it keeps coming back invalid for bambu.

LumiereMarine, 2025-02-04:
This is what I have been using AI so far making. 

[x1c-bricklayers-enhanced.txt](https://github.com/user-attachments/files/18664310/x1c-bricklayers-enhanced.txt)

MagicBOTAlex, 2025-02-04:
Thank you. Weird huh.
I know that these printers definitely can do non-planar prints, since I have custom star g-code that makes a pull-tab for the start wipe.

Idk, but thank you for your response. I'll close this issue now.

LumiereMarine, 2025-02-04:
Comparing the differences of a bambu sliced gcode, and a sliced gcode on prusa of bricklayer, I asked AI about the differences and got this: 

Key Differences and Considerations

Comments and Markers: The Prusa G-code uses comments like ;TYPE:Internal perimeter and ;TYPE:External perimeter to mark wall sections, while the Bambu G-code uses ;FEATURE: Outer wall and ;FEATURE: Inner wall. 

Layer Change: Prusa G-code uses ;LAYER_CHANGE to denote layer changes, while Bambu G-code uses a combination of ;LAYER_CHANGE followed by a line with ;Z: to indicate the new layer height. 

Wipe and Z-Hop: Bambu G-code includes specific wipe sections marked by ;WIPE_START and ;WIPE_END, and it also uses Z-hops during travel moves. 

Extrusion Mode: Bambu G-code uses relative extrusion mode (M83), while Prusa G-code might use absolute extrusion mode (M82). 

Header and Configuration: Bambu G-code has a more extensive header and configuration section.

MagicBOTAlex, 2025-02-05:
any line with a semi-colon marks the rest of the line as comments. This does not affect the printer action. (Hopefully, since Bambu is not open with theirs implementation.

But from my experience making g-code for my Bambu, it is the same.

This means in a line:
'g28; home'
Then "g28" is the command and rest is ignored.

If no command given, then that line is just a commnet.

But again, I can't see why this would actually cause a problem, since the generated g-code's header is usually just a shit ton of comments.

I'm going to re-open this issue, because I want to see what others think.

xorinzor, 2025-02-09:
@LumiereMarine just going to warn you that trying to code something with AI when you don't understand the programming language can lead to dangerous results, including damaging your printer. Please be really careful with these kind of things.

AI might be a fun chatbot, but for coding they are still severely lacking proper reasoning and coding capabilities, and should not be relied upon.

dvomang, 2025-02-24:
> Thank you. Weird huh. I know that these printers definitely can do non-planar prints, since I have custom star g-code that makes a pull-tab for the start wipe.
> 
> Idk, but thank you for your response. I'll close this issue now.

Can you share the gcode? That sounds awesome!

MagicBOTAlex, 2025-02-24:
> > Thank you. Weird huh. I know that these printers definitely can do non-planar prints, since I have custom star g-code that makes a pull-tab for the start wipe.
> > 
> > Idk, but thank you for your response. I'll close this issue now.
> 
> @MagicBOTAlex Can you share the gcode? That sounds awesome!

From line 309 to line 314
https://github.com/MagicBOTAlex/ZSwitch/blob/master/gcode/AlexBetterStartup.gcode

This script is what I use. I have a P1S. So I have no idea if it works on other printers.

MagicBOTAlex, 2025-02-24:
And for the rest of the issue. I'm pretty sure that Bambu is supported. I saw someone in a Reddit comment make a guide. 

So closing the issue as Bambu should be supported somewhat.

But honestly, switching to Orca would probably be easier and safer.

closed 2025-02-24 by MagicBOTAlex

MagicBOTAlex, 2025-02-24:
Update: I switched to Orca, and it is remarkable how seamless the transistion was.
I just log-in with my Bambu account, and all my presets and settings are automatically imported.