TengerTechnologies/Bricklayers #59: Weird layers gaps in Orcaslicer

rgauvin630, 2025-02-09:
I'm getting strange layer gaps in Orcaslicer with the script. Anyone have an issue like this? 

3 wall layers
1 bottom layer
1 top layer

[image]

jurassicjordan, 2025-02-10:
ive also been having similar gaps

SlavicEwok, 2025-02-16:
I have the exact same issue, but even more pronounced. I actually have these gaps on nearly every layer.

DiMENZi0N, 2025-02-24:
Watching this. Having this same issue with Orca 2.2.0 slicing.

IguanaBob, 2025-04-04:
Same on 2.3.0 Linux version. Build 70931e5

[image]

IguanaBob, 2025-04-04:
Did some testing, I notice, at least with my setup, if I use Arachne some of the Z changes don't show lines starting with G1 Z.... which is what the script is looking for. This is in the file directly from Orca before I run it through the script, so not a fault of the script. Not sure if it's a klipper thing, a problem with Arachne, or something else, but if I turn of Arachne and use Normal wall generator things improve, but it's not eliminated.

[image]

IguanaBob, 2025-04-04:
More info:

Some of the Z changes happen mixed in other lines. Here is the raw gcode before running through the script:

G1 Z11 ; restore layer Z
G1 Z12 F7200 ; normal lift Z
G1 X105.619 Y114.734 **Z12** ; move to first infill point
G1 Z11 ; restore layer Z
G1 Z12 F7200 ; normal lift Z
G1 X102.555 Y114.792 **Z12** ; move to first infill point
G1 Z11 ; restore layer Z
G1 Z12 F7200 ; normal lift Z
G1 X100.914 Y110.437 **Z12** ; move to first infill point
G1 Z11 ; restore layer Z
G1 Z12 F7200 ; normal lift Z
G1 X102.486 Y111.857 **Z12** ; move to first infill point

Looking to see what else can be done.

IguanaBob, 2025-04-04:
Changing line 55 got rid of the biggest gaps with Arachne enabled but still doesn't eliminate the small gaps regardless of if Arachne is on or off.

        if line.startswith("G1 Z"):

changed to

        if " Z" in line:

It fixes one problem at least. I'll PR it for that at least.