TengerTechnologies/Bricklayers #60: Linux Support

naytmare1, 2025-02-10:
With Linux coming with python already how would the post processing script look like?

vaut, 2025-02-10:
"/usr/bin/python3" "/Path/To/Script/bricklayers.py" -layerheight 0.2 -extrusionMultiplier 1.3

nickehallgren, 2025-02-10:
Works for me in linux, but -layerheight should be -layerHeight

naytmare1, 2025-02-10:
Awesome will try that and see if it works thanks so much

naytmare1, 2025-02-10:
Tried it, doesnt seem to be working on mine either

nickehallgren, 2025-02-10:
Remove the quotation marks,
 /usr/bin/python3 /Path/To/Script/bricklayers.py -layerHeight 0.2 -extrusionMultiplier 1.3

naytmare1, 2025-02-11:
tried it again, getting a "not directory" error

nickehallgren, 2025-02-11:
Where did you download the script? In my example it is in Downloads and you can check that it is really there from the terminal like this:

`ls /home/username/Downloads/bricklayers.py`

the output should then be:

/home/username/Downloads/bricklayers.py

if the file is not there you get:

ls: cannot access '/home/username/Downloads/bricklayers2.py': No such file or directory

When you know where the script is you can try this in the terminal:

`/usr/bin/python3 /home/username/Downloads/bricklayers.py`

the output should be:

usage: bricklayers.py [-h] [-layerHeight LAYERHEIGHT] [-extrusionMultiplier EXTRUSIONMULTIPLIER] input_file
bricklayers.py: error: the following arguments are required: input_file

Now you are ready to use it in your slicer like this:

`/usr/bin/python3 /home/username/Downloads/bricklayers.py -layerHeight 0.2 -extrusionMultiplier 1.3`

nickehallgren, 2025-02-11:
try python3 and use quotation marks on the path as it has spaces in it :

`/usr/bin/python3 "/home/deck/Documents/3D Print Script/bricklayers.py" -layerHeight 0.2 -extrusionMultiplyer 0.99`

naytmare1, 2025-02-11:
[image]

Im getting this error now, i tried using just python(not python3) still the same

nickehallgren, 2025-02-11:
The error message tells you what is wrong, unrecognized arguments -extrusionMultiplyer it should be -extrusionMultiplier (in my last post I just copied your earlier line before adding quotation marks so it was wrong, but now it has been corrected)

naytmare1, 2025-02-12:
It worked this time now, the main fault was my faulty English grammar thank you so much for the help the quotation mark on the script location helped too wouldnt have known about that at all, can I know the reason why only keeo the quotation mar on the script instead of both?

nickehallgren, 2025-02-12:
You need them because there are spaces in the path of the script:

`/usr/bin/python3 "/home/deck/Documents/3D Print Script/bricklayers.py"`

if your folder would be named 3D_Print_Script you don't need quotation marks (spaces replaced with _)

`/usr/bin/python3 /home/deck/Documents/3D_Print_Script/bricklayers.py`

this also works but is not needed

`"/usr/bin/python3" "/home/deck/Documents/3D_Print_Script/bricklayers.py"`

naytmare1, 2025-02-12:
Ahh i see what you mean by spaces now, i appreciate all the help thank you so much, thiz issue is resolved and im closing again thanks sk much

closed 2025-02-12 by naytmare1