TengerTechnologies/Bricklayers #63: WeaveLayers open standart?

JPluess, 2025-02-11:
Description:

I am proposing the development of a new weave layer design featuring an interlocking wave technique, illustrated in the attached image. This design incorporates both 'Z' and 'Y' wave patterns, which serve as a basis for the structure. While the gaps within the pattern need to be filled (though they are not visualized in the image), the parameters can be adjusted to create various wave forms, including zigzag or trapezoidal movements.

The outer walls of the weave remain untouched, allowing us to focus on the intricate inner weaving design. This project aims to explore the creative potential of weaving "to the heart's content," fostering innovative and open-source contributions from the community.

Please check the attached illustration for reference! Looking forward to your thoughts and ideas.

[image]

pilot51, 2025-02-12:
Great idea in theory, but I'd want to see a proof of concept (real-world results) before I believe it would work with acceptable quality. With nozzles having a flat bottom more than double the extrusion diameter, I would expect it to collide with neighboring lines that are above the extruding line. It has the potential to be very messy.

I assume the outer perimeters are printed last so they're always at the lowest height of the weave layer being printed and not a collision risk.

SnakeLayer: I think the main issue there is the higher part of the black being smooshed by the nozzle when it dips down. I doubt it would be a smooth wave, certainly not consistent thickness.

TrapezLayer: Green and black have a wide conflict area when it dips below the height of one already printed. The nozzle would tear up the high point of the neighbor for the entire length of the dip.

HoneyLayer: The area of conflict appears to be where it changes height next to the same color. The nozzle would dig a little into the already printed neighbor of the same color near where they come together.

dblanchard, 2025-04-11:
I don't yet have the 3D modeling skills to show what I'm thinking. I think all the ideas have great merit, but @pilot51 raises excellent concerns. I think instead of moving the extruder up and down, cf. the SnakeLayer, you could adjust the volume of filament being extruded. Looking at the SnakeLayer diagram, yellow, blue, and red would all be the standard flow, as would the high points (the peak of each wave) for black, but the low points (the trough of each wave) would maybe double the flow while keeping the extruder at the same z- value as at the low points. I do think what you have illustrated would be stronger than what I'm suggesting. I just think the physical limitations of the nozzle head prevents what you have shown.