TengerTechnologies/Bricklayers #70: stringing issue

RaceyDay, 2025-03-09:
theres something in the script that causes stringing or oozing on prints. i got past the error 2 an the solid blocks i was tsting on looked great and i could tell a strength difference but once i triedprinting something with shaped walls and important tolerance internals there is stringing that is very hard to tweak away with retraction setting and a few other vaiables. but that results in the layers not being bricked any more, same settings but without the post script results in a flawless print.

RaceyDay, 2025-03-09:
tested on prusa slicer and orca slicer. results are the same with both. prints solid objects fine but with any internal hollow parts of the object has stringing, little to no stringing from part to support material

ezxzeng, 2025-03-18:
I've also noticed this. there's a diagonal line across my internally hollow box.

[image]

GeekDetour, 2025-03-19:
Thanks @ezxzeng for posting this photo. I've been playing with another implementation of BrickLayers and in the beginning I was having A LOT of stringing. I implemented my own wiping routines... It got considerably better... but not completely clean yet. There might be something more challenging about it than what we anticipated.