TengerTechnologies/Bricklayers #74: Error Code: 2 When trying to generate NonPlanarInfill

SeeTurtlee, 2025-04-20:
[image]

Hi there! I've been trying to get some gcode to generate using this post processing script. However, i keep getting the error as shown in the image. I've tried installing and reinstalling multiple diffrent versions of python, adding there paths to system variables, and messing with the script arguments. All yielding the same error. Even the example arguments paste exactly as they are, won't work.  I've also moved the script location around to see if that would work. Right now, i'm out of things to try. (I got my python location from "where python" and everything looks fine in that location.

I manually ran the script on some gcode i generated to get a z_shift log (not sure how to add arguments to it via cmd, i did try though) . not sure if this can help at all. 

[z_shift_log.txt](https://github.com/user-attachments/files/19827446/z_shift_log.txt)

I appreciate all the work put into this and i can't wait to see what OP comes up with next!!

SeeTurtlee, 2025-04-20:
False alarm. apparently all i needed to do was make this post. i was using the arguments for non planar interlocking walls... brain fart :P

closed 2025-04-20 by SeeTurtlee

TengerTechnologies, 2025-04-20:
If you are here of todays video, you should use the [NonPlanarInterlockingWalls.py]
And if so, make sure to do -include-external-perimeters instead of -include-external-perimeter