TengerTechnologies/Bricklayers #75: Scripts Not Working (All slicers)

SeeTurtlee, 2025-04-20:
Hi, 

I've been working to get this script working for a bit. at first i was getting an error due to incorrect arguments. But now, my generated slice doesn't seem to be altered in any way. python seems to be working as if i change the argument to something that can't be read, it will error. I also use python for other things and it works just fine. i get my path from where python. i've tried 3.13 and 3.12. I've tried all 3 scripts and the example arguments. It slices just fine, and the cmd pops up for a second when it gets to the post-process script. just nothing is changed. as far as i can see as its up for a split second, the cmd says "DEBUG: using lookup table for (slicer name)". Any help with this would be greatly appreciated!!

i made sure to follow the instructions such as, arc fitting off and bgcode off. (the main one i'm trying to use is NonPlanar Interlocking Walls)

[image]

fang64, 2025-04-21:
I'm seeing the same thing in Bambu Studio v1.10.2 and v2.0.2.58 releases 

[image]

fang64, 2025-04-21:
As a follow up this is prusaslicer,

[image]

fang64, 2025-04-21:
@SeeTurtlee were you able to get it working?

SeeTurtlee, 2025-04-26:
> [@SeeTurtlee](https://github.com/SeeTurtlee) were you able to get it working?

Yes and no. I was able to  get one to "work". Well sort of. It ended up somehow getting a slice that took the benchy and mashed it down to like 60 layers, could have been more as when i went back and looked at the preview, it looked like some spots had very thin pillars that could have gone to the original height. i tried to slice it again and overwrote the file with the same exact settings, and it didn't work. not sure why. i would have loved to post that here to be looked at but its gone :(. 

Also, It actually sliced the back half partially below the plat on some of the waves, not exactly sure how that happened, i stopped it when i noticed the nozzle bending. Luckily it caused minimum damage to the plate and nozzle, just a few specs when it touched.

fang64, 2025-04-27:
Interesting, so you are able to produce gcode that does something, I can't even get it to make changes to the input. It just spits out ;not processed comments all the way down.

SeeTurtlee, 2025-04-27:
Well, that's what it's doing to me again now. with nothing changed. Supper confused as to why.

AxE-pl, 2025-05-17:
I have same issue, What gpt proposed to me is to use fork of bricklayers by GeekDetour. And it worked, but still didn't find way to use nonplanar walls

AxE-pl, 2025-05-17:
to OP: try turning off scarf joint if you have it on and turn zhop to at least 0.1

DasCarstel, 2025-05-20:
I successfully ran the `NonPlanarInterlockingWalls.py` script using the following command:
```
C:\Users\carst\AppData\Local\Programs\Python\Python311\python.exe C:\Users\carst\Downloads\NonPlanarInterlockingWalls.py -include-infill -infill-amplitude 0.3 -include-external-perimeter -infill-frequency 2 -include-perimeters -infill-direction y -wall-direction xy -max-step-size 0.1 -wall-amplitude 0.4 -wall-frequency 2
```
I found on [this GitHub issue](https://github.com/GeekDetour/BrickLayers/issues/20) that I needed to use the G-code output. Before running the script, I made sure that Arc fitting was deactivated. After running the script, I opened my Bambu Lab slicer software and used the `File -> Export -> Export G-code` option. Subsequently, I dragged and dropped the generated `Vesa_PLA_1h45m.gcode` file into the slicer. The image below shows that this process was successful.

[image]