TengerTechnologies/Bricklayers #76: Prusa Slicer: no way to get rid of Error 1

MauroLXX, 2025-04-23:
Hi,
I have tried (one at a time) all the .py scripts.
I have tried them with Prusa Slicer 2.9.2, running under Python 3.13.3

The situation is as follows:

- bricklayers.py works correctly
- bricklayersNonPlanarInfill.py works correctly
- **NonPlanarInterlockingWalls.py** does not work (Error code: 1)
- vertical_bricklayers.py does not work (Error code: 1)

By "works correctly" I am not making any judgements on the quality of the output generated, but only on the fact that the script is executed correctly.

I was most interested in the “**NonPlanarInterlockingWalls.py**” script, but unfortunately, I could not get it to work at all.
- I disabled the BGcode from both the PS settings and the print profile settings;
- I disabled ArcFitting, checked the paths and file names letter by letter;
- I tried putting the file names all in lower case;
- I tried running the script with and without additional parameters.

The paths to Python and the scripts are set correctly because, otherwise, I would receive Error code: 2

I have no more ideas to apply.

Any further suggestion would be very appreciated.

Thank You

[image]

[image]

[image]

lestatclemmer, 2025-04-29:
Did you ever figure this out? I'm having the same issue it seems

MauroLXX, 2025-04-30:
The short answer is “no”.

I spent a whole afternoon trying to figure out if there was some syntax error or some wrong upper or lower case letters.

I checked and rechecked all the folder paths, even going so far as to move the files directly to “c:/”.

Always and only Error code 1...

Finally I gave up and tried using ‘bricklayersNonPlanarInfill’ as a fallback alternative.

‘bricklayersNonPlanarInfill’ works from a Python point of view because it does not create any errors, but unfortunately, in the end, this is not productively usable either.
The reason is that slicing processing generates GCode with numerous anomalous paths, even on simple objects like the one in the attached screenshot.

[image]

[image]

lestatclemmer, 2025-05-02:
Well that's unfortunate. I've been able to get NonPlanarInterlockingWalls.py to work by running it in a command line with an already-created .gcode file. Not the cleanest method (relative to having it integrated with Prusa Slicer,) but it works fairly alright from what I can tell. I'm still tweaking the parameters and playing with it.

gingerbreadassassin, 2025-08-06:
Try running prusaslicer as administrator. 

When running as non-administrator:
```
c:\Program Files\Prusa3D\PrusaSlicer>prusa-slicer-console.exe
[2025-08-06 12:34:50.679130] [0x00001644] [error]   Failed to load credentials from the system password store.
Traceback (most recent call last):
  File "C:\Users\meep\Downloads\NonPlanarInterlockingWalls.py", line 82, in <module>
    logging.FileHandler("gcode_debug.log"),
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python313\Lib\logging\__init__.py", line 1218, in __init__
    StreamHandler.__init__(self, self._open())
                                 ~~~~~~~~~~^^
  File "C:\Program Files\Python313\Lib\logging\__init__.py", line 1247, in _open
    return open_func(self.baseFilename, self.mode,
                     encoding=self.encoding, errors=self.errors)
PermissionError: [Errno 13] Permission denied: 'c:\\Program Files\\Prusa3D\\PrusaSlicer\\gcode_debug.log'
[2025-08-06 12:35:30.099762] [0x00004c78] [error]   Post-processing script "C:\Program Files\Python313\python.exe" "C:\Users\meep\Downloads\NonPlanarInterlockingWalls.py" -include-infill -include-external-perimeter -include-perimeters -alternate-loops on file C:\Users\meep~1\AppData\Local\Temp\.19128_0.gcode.pp failed.
Error code: 1
```