TengerTechnologies/Bricklayers #78: Bambu - Non-Planar Interlocking walls - Not working

alenoi, 2025-05-04:
So it should work with Bambu but it just makes a mess unfortunately.
Missing and/or messed up walls, extra wall lines instead of movement, and messed up infill (adaptive cubic)

BambuStudio 2.0.3.54

Used command:
"C:\Users\Tomi\AppData\Local\Programs\Python\Python312\python.exe" "D:\Documents\3D_print\_PostProcess\Bricklayers\NonPlanarInterlockingWalls.py" -include-infill -infill-amplitude 0.3 -include-external-perimeter -infill-frequency 2 -include-perimeters -infill-direction y -wall-direction xy -max-step-size 0.1 -wall-amplitude 0.4 -wall-frequency 2

[image]
[image]
[image]
[image]

BBrenner2217, 2025-05-23:
I got it working with my H2D after a lot of trial and error. I forgot exactly which setting was holding me back but I will take a look later and post back here.

Here's a video of it working, printing nylon. 
https://youtube.com/shorts/o5qnBsN-axs?si=inpHTGSki7Q0nhF8

TengerTechnologies, 2025-10-30:
@machinaexmente I'm not sure but try disabling arc moves. I belive in Bambustudio it is called arc fitting in the Quality tab. I can have a look later today if it is not that setting